PaulStoffregen/ILI9341_t3 #7: private

FrankBoesing, 2015-04-30:
Hi Paul,
all low-level functions are private, which makes it impossible to use them for own fast functions without changing the library.  For example, for my spectrum-bargraph display it would be good to have an own specialized fillrect function - this is not possible without changing the library code, now.

What do you think ? Would it be possible to remove the "private" ?

PaulStoffregen, 2015-04-30:
Would changing from private to protected be enough?  Or do they need to become public?

FrankBoesing, 2015-04-30:
Hi Paul,
i think protected is ok :-)

Thank you !

PaulStoffregen, 2015-05-05:
done https://github.com/PaulStoffregen/ILI9341_t3/commit/136b80bea04641277b9df389010b312aedb1734a

closed 2015-05-05 by PaulStoffregen