PaulStoffregen/ILI9341_t3 #17: Smoothscrolling: Printing fonts line-by-line

FrankBoesing, 2015-11-18:
It would be cool to have smooth-scrolling
This requires a way to print the fonts per (pixel-)line.
In addition, a way to determine the screen-width of string would be needed to detect newlines

blackketter, 2015-11-19:
I have working code for measuring text width and height as well as wrapping that I'll submit as pull request which should make this easier.

blackketter, 2015-11-19:
BTW: One way to do the per-pixel rendering of fonts would be to set a clip rectangle for just that line and then render away.  It wouldn't be particularly optimized, but the clipping would prevent much of the actual drawing from happening. See https://github.com/PaulStoffregen/ILI9341_t3/pull/13

closed 2017-09-14 by FrankBoesing