PaulStoffregen/ILI9341_t3 #23: Missing Symbols using Arduino IDE 1.6.7

waldo1979, 2016-03-02:
Getting some compilation errors while attempting to compile in 1.6.7 due to missing symbols:

ILI9341_t3/ILI9341_t3.h:178:9: error: 'KINETISK_SPI0' was not declared in this scope
ILI9341_t3/ILI9341_t3.h:193:31: error: 'SPI_SR_TCF' was not declared in this scope
ILI9341_t3/ILI9341_t3.h:197:67: error: 'SPI_PUSHR_CTAS' was not declared in this scope
ILI9341_t3/ILI9341_t3.h:197:71: error: 'SPI_PUSHR_CONT' was not declared in this scope

Any idea where these should be declared?

waldo1979, 2016-03-02:
Fresh install of Windows 10, attempting to compile the graphicstest example.

blackketter, 2016-03-02:
Make sure you have Tools > Board set to a Teensy 3.x

closed 2016-03-02 by waldo1979

PaulStoffregen, 2016-03-02:
@blackketter - Thanks!