PaulStoffregen/ILI9341_t3 #37: ESP8266 Hardware causing compile error due to missing KINETISK_SPI0, SPI_SR_EOQF, etc

gojimmypi, 2016-11-07:
Greetings - 

I was hoping to use this cool library with the ESP8266 from the Arduino environment (I am using Visual Micro in Visual Studio). The problem I am having is related to a missing declaration (perhaps for different target hardware?) at compile time:

error: 'KINETISK_SPI0' was not declared in this scope
error: 'SPI_SR_EOQF' was not declared in this scope
error: 'SPI_SR_TCF' was not declared in this scope
(and then a ton of related errors)

Can the ESP8266 hardware be used with this ILI9341_t3 library?

thanks

PaulStoffregen, 2016-11-08:
> Can the ESP8266 hardware be used with this ILI9341_t3 library?

No, it's not supported.

closed 2016-11-08 by PaulStoffregen

WA9SGD, 2020-12-30:
I'm trying to use ILI9341_T3 with an STM32F1. I'm getting similar errors to the previous post. Is this supported?
Is this library specific to Teensy?

FrankBoesing, 2021-01-04:
Yes it is. It will not work on other CPUs.