PaulStoffregen/ILI9341_t3 #42: Using your enhanced library with other chips

marcmerlin, 2017-02-18:
Hi,
I found your library on
https://forum.pjrc.com/threads/26305-Highly-optimized-ILI9341-(320x240-TFT-color-display)-library?s=0f903ddac2e0f07c2725be15c8738a95
although I have an ESP32 with which I can't get your library to build, mostly due to KINETISK_SPI0 which doesn't exist on the ESP32 arduino emulation layer.
I don't actually really need your faster library that much since ESP32 has a fast ILI9341 already, but I wanted to use your library for the extra functions you've added, and more importantly the new demos you have like pictureEmbed.ino and DemoSauce which don't exist in the adafruit library and won't work with it due to lack of methods in there.
Unless you have a better idea, I think my best bet is to make your library work on ESP32. Would you have a suggestion of how I can replace/revert KINETISK_SPI0 into something more arduino compatible?

blackketter, 2017-02-18:
I've been thinking about this problem a bit too, but wonder if it might be better to port the new ILI9341_t3 APIs and Teensy optimizations back to the original Adafruit library which supports a large number of CPUs and displays.  

@marcmerlin - Which ESP32 ILI341 are you already using?

marcmerlin, 2017-02-18:
I'm using the adafruit lib with hw SPI. It works but I'd love to be running
your version since it supports better example code. Is it easy to convert
the special KINETISK_SPI0  calls you're making into normal hw SPI calls ?
 If so, I could give it a shot.

PaulStoffregen, 2017-02-20:
No, you can't use KINETISK_SPI0 on any ESP chip!  It's very specific to the SPI hardware Freescale designed in their Kinestis chips.

marcmerlin, 2017-02-20:
I understand KINETISK_SPI0 will not work, what I was asking is can you describe what I should replace it with, which will work with normal hardware SPI on other chips (yes, it'll be slower, but I'd rather use your enhanced lib a little bit more slowly than not at all :)

marcmerlin, 2017-02-21:
Obviously if I get it working on another chip, I'll send you a PR you can hopefully merge and then you won't have to worry about doing this yourself and more people will be able to benefit from your code.

marcmerlin, 2017-02-21:
(replacing the previous message that was not very clear)
I understand KINETISK_SPI0 will not work, what I was asking is:
can you describe what it does so that I can replace it something more generic which will work with normal hardware SPI on other chips (yes, it'll be slower, but I'd rather use your enhanced lib a little bit more slowly than not at all :)
Or if you know what I can drop in in its place, even better :)

PaulStoffregen, 2017-02-21:
These SPI registers are documented in the reference manual for the chip.  The KINETISK_SPI0 fields simply represent these registers.

https://www.pjrc.com/teensy/datasheets.html

I suggest you look at the MK20DX256 reference manual, chapter 45.  The register map is section 45.3, starting on page 1123 and the detailed documentation is section 45.3.1 starting on page 1126.

There's also a few key details in chapter 3, section 3.9.3, starting on page 127.

marcmerlin, 2017-03-04:
Thanks for your answers Paul, it looked kind of non trivial and for now I really only needed a single bit, which was pixmap display, so I ported it back to the adafruit library:
https://github.com/espressif/Adafruit_ILI9341/pull/1

marcmerlin, 2017-03-05:
Great news, I was able to port the missing functions from your lib to the adafruit lib and get demosauce working (well, mostly, some demos crash on my board, but I think my board has a power problem).  Updated pull request (not for your lib): https://github.com/espressif/Adafruit_ILI9341/pull/1
See a few screenshots and videos: https://goo.gl/photos/bMVqdiAxDkEppuN88
[image]
[image]

marcmerlin, 2017-04-05:
Ok, and between me-no-dev and I, we've ported enough of your library back to the adafruit ILI lib, that things work (including scrollto and displaybitmap).
This doesn't port your library back to other chips, but after looking at it, I realized that making it work on ESP32 or other chips would just be too painful and that it'd be quicker to port its extra functionality back to the adafruit lib.
I don't have every demo from demosauce working, it looks like some might be buggy, but I have enough working.
https://www.youtube.com/watch?v=OmOlm4Y6Clk
As a result, I'll close this issue.

closed 2017-04-05 by marcmerlin