PaulStoffregen/ILI9341_t3 #58: what is KINETISK_SPI0.SR?

dduehren, 2020-05-23:
Please use this form only to report code defects or bugs.

For any question, even questions directly pertaining to this code, post your question on the forum.

Teensy: forum.pjrc.com

If you are experiencing trouble but not certain of the cause, or need help using this code, ask on the forum.  This is not the place to ask for support or help, even directly related to this code.  Only use this form you are certain you have discovered a defect in this code!

Please verify the problem occurs when using the very latest version, using the newest version of Arduino and any other related software.

----------------------------- Remove above -----------------------------

### Description

Describe your problem.
When I try to run an example program on a Nano 33 IOT, the compiler complains about all the references to KINETISK that are not defined.  Is the else clause at all tested or valid?

### Steps To Reproduce Problem

Please give detailed instructions needed for anyone to attempt to reproduce the problem.
Install the library
Try to compile graphicstest example
Look at compiler output.
I know this happens with an Nano 33 IOT as the source.  I'm trying to use the xpt2046 in 8 bit mode, but that has nothing to do with the library behavior.

### Hardware & Software
As mentioned, I'm trying to get this to run on a Nano 33 IOT 

Board
Shields / modules used
Arduino IDE version
Teensyduino version
Operating system & version
Any other software or hardware?

### Arduino Sketch

```cpp
// Change the code below by your sketch (please try to give the smallest code which demonstrates the problem)
#include <Arduino.h>

// libraries: give links/details so anyone can compile your code for the same result

void setup() {
}

void loop() {
}
```

### Errors or Incorrect Output

If you see any errors or incorrect output, please show it here.  Please use copy & paste to give an exact copy of the message.  Details matter, so please show (not merely describe) the actual message or error exactly as it appears.

FrankBoesing, 2020-05-23:
It runs on a Teensy 3.x or 4.x only (NXP ARM Cortex-M cpu)

dduehren, 2020-05-23:
Hi Paul,

1. The Nano 33 IOT is a Cortex M0 32 bit SAMD21
2. Then why the else clause?  I may add a define to include my processor and see if it works.  I’ll look at the differences between the Teensy.

David Duehren

PaulStoffregen, 2020-05-23:
This library is for Teensy only.

The code is open source, so you may create a fork and modify your fork any way you like.  But your changes for non-Teensy boards will not be merged here.  Likewise, this is not the place to ask for help to edit the code for another board.

closed 2020-05-23 by PaulStoffregen